Bambu-Research-Group/RFID-Tag-Guide #4: Is there any progrss ?

michalss, 2023-08-01:
Hi,

Is there any progress please ? I would love to get it workig...

Regards

phranck, 2023-09-15:
Yeah, would like to see any progress, too. Reading and writing custom RFID tags would be really awesome.

mplacona, 2023-10-02:
Would love to know this too! Can't wait to have RFID tags on all my spools

warbarz, 2023-10-06:
Hello all!
The rfid tag included with spool also maintains spool consumption. Figuring out the decoding will be essential in reuse of tags on various spools of varying consumption. 
New refills include tags as well for replacing the old tags. This will be instrumental in use of 3rd party or self printed spools. 
What we are doing here will develop into a tool to easily choose type, color, and spool weight remaining to generate a file to write to the rfid tags for detection.
Also suggesting potentially adhearing the tag to the exterior of the spool to ensure read-ability and make writing the tag easier.

I have gathered all the following:

Spools sold by Bambu on their site including their hex identifier posted. That is type, color, and hex code.
I have purchased one or more of each spool as well as a number of refills to compare data against.
I have a proxmark and prepared the environment to begin reading the tags, and intercepting communication with reader.

I am in need of assistance with the following:

Decoding the crc and keys from the dumps of the rfids.
Creating a application to generate dumps to write to the tags based on the progress of the above decoding.
Potentially a ams system that is for parts to harvest the rfid reader and cabling to make reading and intercepting communication between it and spools rfids easier. (If someone has one they have scrapped let me know so we can get the circuit board and rfid reader from it.)

Let me know if anyone has anything to add to the above or if we should start a discord group to more proactively (and privately) communicate on progress.

michalss, 2023-10-09:
I do have Proxmark3 and ASM, but most valuable thing is that im developer i can do mobile, desktop applications (Maui c#, flutter, c++, java), also can do webs(Js,Blazor, React, Svelte)... Let me know how can i help.. There is also possibility to established the annonym server to host this data, ot at least api for Desktop application....

tiagojanz, 2023-10-10:
Hi, I can have a try at the CRC cause I have a lot of experience with it, if you post a complete hex output I can try reverse engineer it.

tiagojanz, 2023-10-10:
yes, search for tiagojanz

warbarz, 2023-10-12:
I sent a friend request. Could not message due to your settings. 😊

doudar, 2023-10-19:
What about dumping the firmware from the AMS? Possibly that would shed light on the decryption process?

warbarz, 2023-10-19:
I have the ability to do chip off read dumps and am willing to do the solder work to remove and read the chip and replace the chip onto my unit. However, unless there is a terminal interface or some way of communicating with the chip through a firmware upgrade/recovery or something, decoding the binary and assembly of a bin file is out of my expertise.

 

 

Yes, having both sides of the process decoded and referenceable will significantly improve our chances of manipulating and making custom firmware/software to leverage the AMS… Imagine the uproar if we made the AMS USB compatible and could use it with ANY printer?! That is likely territory I do not wish to tread on due to legality and rights issues. 

 

But, updating the nfc tag for spool consumption and color change? Sorry if I feel that is square in the rights of the consumer to be able to manipulate even if it does effect a “small” amount of sales from bambu filament.

redosk, 2023-10-21:
> yes, search for tiagojanz

Hi, I've added you also on Discord, I'm interested to join in.
Maybe we should make a server to discuss altogether ?

--

Regarding spool consumption, it is not written on the tag ? Consumption is kept on the AMS, I suppose, right ?

michalss, 2023-10-22:
> yes, search for tiagojanz

also add your to discord...

kolodus, 2023-10-24:
did the discord happen? interested in what others have learned and happy to lend a hand w/ a logic analyzer or scope if useful, or hunt down uart/jtag/spi if i can source a parts only ams

raleighlittles, 2023-11-03:
Has anyone started working on an automated key retrieval process? Specifically this part from the documentation: https://github.com/Bambu-Research-Group/RFID-Tag-Guide#getting-the-other-keys-by-analyzing-the-log-file

@warbarz  ?

warbarz, 2023-11-03:
I have joined one person in discord to discuss who has a good grip on the proxmark stuff. He has posted his discord I believe.

 

I myself have not gotten to dumping bins for him yet. I have all the bambu spools, colors, types, etc. I have a proxmark and 2 ams units. I just havnt had time to dedicate to this between using the printer, working, and family.

 

I would like to work directly with everyone in a discord chat so I can be more useful and perform specific requested time limited functions, like a bulk dumping of the spools I can do over time and post the results as I go… but I need to be told how and what to produce.

redosk, 2023-11-03:
> Has anyone started working on an automated key retrieval process? Specifically this part from the documentation: https://github.com/Bambu-Research-Group/RFID-Tag-Guide#getting-the-other-keys-by-analyzing-the-log-file
> 
> @warbarz ?

I've made some scripts.

Also, I just created a Discord server, it will perhaps be easier to discuss this : https://discord.gg/K7eqWwUs

michalss, 2023-11-16:
Discord invite is not valid anymore :(

> I've made some scripts.
>
> Also, I just created a Discord server, it will perhaps be easier to discuss this : https://discord.gg/K7eqWwUs

Discord invite is not valid anymore :(

warbarz, 2023-11-16:
https://discord.com/invite/Pk9pPbqBSent from my Verizon, Samsung Galaxy smartphone

pelrun, 2023-12-02:
Invite link has expired again (Discord why u do this?)

closed 2024-12-05 by queengooborg