Bambu-Research-Group/RFID-Tag-Guide #5: RFID Testing

ukdavewood, 2023-08-07:
I imagine this behaviour is pretty obvious - but thought I would document it anyway as it might be of interest.

Test1 - Empty BL PLA White Basic -> Generic White.

- Respooled about half of some generic white PLA onto an Empty BL PLA Basic white spool inner.
- Printer detected it ok as white PLA and also seem to get the approximately amount respooled about right too - so probably handy for auto colour selection of non Bambu filament - as long as you have a previously used Bambu inner, and don't mind the generic filament being picked up as Bambu.
NB. I marked an arrow on the cardboard of the inner to make sure I wound the filament on in the correct direction

Test2 - Removed RFID from left hand side.

-  As expected the respooled generic white filament on a BL spool wasn't detected as white in Slot 4 of the AMS.
- But in slot 3 of the AMS it was detected ok as white.

Test 3 - Removed RFID from left hand side of BL Generic Blue

-   As expected same behaviour as above - the spool was detected ok when it was replaced in slot3

Test 4 - probably the most useful one to me.   

Took a half of the BL Generic Blue filament (which I was already using in another AMS on another printer),  - but having to manually configure it on that printer.

Added the left RFID removed from the main BL Blue PLA spool into the Half respooled spool - on the outside, with the edge of the RFID about 3.7mm from the edge of the spool.   Stuck it on with sticky tape on the outside.

- Result as expected - in slot 4 of the second AMS (on a different print) - filament was detected ok as half empty BL PLA blue.

So I will probably do the same for the other 4 or 5 BL filaments that I have half respooled onto 2nd spools - so that I can use the same colour on 2 printers at the same time.   Just need to mark the 2 spools carefully to make sure that it is obvious which ones will be detected in the odd and even slots respectively.

The only slight change I might make is that I think it would probably be better to slice off the right hand side RFID - as it won't interfere with the notch - where as the left hand side is the one right on the notch in the spool holder.

Haven't actually tried doing any printing on long term tests - but thought I would document results anyway in case they might be of use.

michalss, 2023-10-09:
I do have Proxmark3 i can also help, but most valuable thing is that im developer i can do mobile, desktop applications (Maui c#, flutter, c++, java), also can do webs(Js,Blazor, React, Svelte)... Let me know how can i help..

Imtherealnoto, 2023-11-03:
Mine have unique tags on each side of the spool that normal?

wzqvip, 2024-11-02:
> Mine have unique tags on each side of the spool that normal?

As far as I know, such as X1C only have NFC reader between 1,2 and 3,4. It can't detect the left tag on 1. These tags should be the same.

closed 2025-08-13 by queengooborg