Bambu-Research-Group/RFID-Tag-Guide #6: mf_nonce_brute finds incorrect keys, leading to incomplete key retrieval

redosk, 2023-10-21:
Sometimes, when you include a key found by `mf_nonce_brute` in the dictionary, trace list will suggest the same `mf_nonce_brute` command line. It's because the key is not the good one.

In our case, when the decoding is successful, the decoded command always starts with '30'. For example :
```
enc:  5315320D
dec:  30200089
```

However, when the decoding is incorrect, the decoded command differs in Phase 2, even though it was correct in Phase 1 :
```
CMD enc( 8f090159 )
    dec( 300426ee )    <-- valid cmd
...
enc:  8F090159
dec:  60D1F1BC
```

So, I edited `mf_nonce_brute.c` and commented out all commands except for the one starting with 0x30, as specified in `protocol.h`, before recompiling it :
```
uint8_t cmds[8][2] = {
    {ISO14443A_CMD_READBLOCK, 18}
/*
    {ISO14443A_CMD_WRITEBLOCK, 18},
    {MIFARE_AUTH_KEYA, 0},
    {MIFARE_AUTH_KEYB, 0},
    {MIFARE_CMD_INC, 6},
    {MIFARE_CMD_DEC, 6},
    {MIFARE_CMD_RESTORE, 6},
    {MIFARE_CMD_TRANSFER, 0}
*/
};
```

Now it decodes with the correct key :
```
CMD enc( 8f090159 )
    dec( 300426ee )    <-- valid cmd
...
enc:  8F090159
dec:  300426EE
```

Also, don't forget to include in the dictionary the keys that are listed as 'WEAK' or as 'nested probable keys' in the trace.

raleighlittles, 2023-11-03:
I have a key-related question that wasn't clear to me from just reading the documentation. 

On this step: https://github.com/Bambu-Research-Group/RFID-Tag-Guide#getting-the-other-keys-by-analyzing-the-log-file

Where it says:

> Enter the keys line by line into that file.

Where do we get the key**s** used? I ran the `mf_nonce_brute` command and only got one key candidate.

Should that be the only line in the dictionary file when I run it the first time?

redosk, 2023-11-03:
@raleighlittles Yes, it's one key at a time.
You add the first key in the dictionary, and you do step 1 again, you will have the next key to crack, do this until you don't have any key to crack.
Don't forget to add also the weak and "nested probable" keys listed in the trace and you should be able to generate the key file at the next step.

Also, keep your dictionary for the next spool, the keys can be reused.

queengooborg, 2024-12-05:
This has been fixed in #40!

closed 2024-12-05 by queengooborg