Bambu-Research-Group/RFID-Tag-Guide #10: Man in the Middel

math-milan, 2023-12-14:
Hi, I would like to propose a solution that may be easier then to create counter fit NFC tags. Maby its a bad Idea but i just wanted to proposed anyway. 

What if we used a mcu as man in the middle from the AMS main board to the rfid reader the man in the middle spoofs all the checks like is it writable ( i thought i heard that this is a problem ). 

In my mind you have the advantage that you can use all compatible RFID tags that work with the chip they use ( FM17580-QNA-T-G [lcsc](https://www.lcsc.com/product-detail/RF-Chips_FUDAN-MICRO-FM17580-QNA-T-G_C2895152.html)), 
but off cores you would need to create the board and hardware so please let me know what you think off this. At the moment I don't have the time to look into it but after my exams at the start off next year I have plant off time for it.

wzqvip, 2024-11-02:
Probably the RFID reader is a general pn532 or something, use serial to communicate and only passes the content of the tag(including the last RSA fingerprint), thus it doesn't make any different to hack them.

Luro02, 2024-11-24:
> What if we used a mcu as man in the middle from the AMS main board to the rfid reader the man in the middle spoofs all the checks like is it writable

Like mentioned in the comment above this one, the biggest problem is the signature. If the signature is validated on the AMS mainboard, it would not be possible to support custom tags in this way. (You need the private key that only bambulab has to sign custom tags, and there is really no way to spoof that without modifying the firmware).

What you could do is replace invalid tags and forward valid ones to the AMS, but it is questionable how useful that is (this effectively the same as you cloning the tag, which won't get you support for different profiles and colors).

-----

I am mainly replying to this issue, because I had a similar thought, while thinking through the "next step" which would be a custom firmware for the AMS.

### Intercepting read tags

Assuming the mainboard for reading the rfid tags is just a board with the reader chip on it that forwards everything to the main ams board, one could build a board that sits in between that connection.

Instead of spoofing/faking stuff, one would continue to forward everything to the main AMS board and just read the contents. If it detects an invalid tag (or maybe some special marker) then it could set the filament like https://github.com/capull0/BambuAMSCtrl here.

The benefit of this would be that it is independent of the AMS firmware, and custom filament tags would just work after one added that man in the middle board

closed 2024-12-24 by math-milan