Bambu-Research-Group/RFID-Tag-Guide #11: Bambulab Firmware for AMS

MultiStorm, 2023-12-19:
I don't believe it, but is the firmware for the X1 or the AMS available for download?
You could check whether the public key is included in the firmware. If the key is contained in the firmware, you could perhaps inject your own key? Unfortunately I have not found a download option for the firmware.

gshpychka, 2023-12-19:
The firmware is not open source.

MultiStorm, 2023-12-19:
i hadn't expected open source either. but i was hoping for a binary file :-)

gshpychka, 2023-12-19:
> i hadn't expected open source either. but i was hoping for a binary file :-)

You wouldn't be able to extract the key from a binary.

MultiStorm, 2023-12-19:
the private key anyway. but maybe you can exchange the certificate if you can access the setup file. i just don't know in which format a new firmware is transferred. maybe you can find out the download path with logs in the network?

CodeBradley, 2024-04-24:
If you have pfsense you can log the packets being sent to and from the lan IP address. I'll try this out

sparrowhe, 2024-11-15:
you can extract the ota infomation from mqtt message. as far as i know, bambu A1mini (without AMS Lite) will send available updates on mqtt, I'm not sure if AMSLite's information is also sent this way.