Bambu-Research-Group/RFID-Tag-Guide #13: Question: Whats neccessary to get further with this Project ?

rauschgiftengel, 2024-01-05:
As metioned in the description - what do you need to get further ? 
I´m really looking for a solution to use my own filament with own tags.

At the moment i own a Bambu Printer with AMS and 4 Spools with Tags - Already tried to sniff with the Chamelon Mini (but i guess sniffing works different here) and with the Chameon Ultra (but at the moment it has no sniffing implemented)

But im more then ready to order a proxmark 3 to play with the ams ;)

So back to the question, my guess is that your approach is to gather as many (partialy)decrypted tags as possible to generate those keys.
And with that keys we can hopefully decrypt the complete tag, modify the data and write it to our own tags ?

warbarz, 2024-01-06:
Join the discord 😊 

 

Big take aways:

We have all spools tags

We can decrypt tags

We have automated the process

We cannot crack the crc to make our own tags due to good encryption used

We are investigating network traffic and custom firmware now

There are some really smart people working on this 😊

Sarah-1331, 2024-01-06:
2 what discord ??

Schokobecher, 2024-01-06:
You can find it here 
https://discord.gg/TnY65WcH

rauschgiftengel, 2024-01-11:
So, warbarz said 

"Join the discord 😊
Big take aways:
We have all spools tags" 

All i can see in the Discord is 4 Dumps, asked there about the tags or dumps, but never got an answer. 
So, do you have all the tags or not ? 

Second question was, if you have all the tags, there should be a list with the keys for the tags - as they are per tag.
Asked about that, still no answer.

I just want do use the dumps, put it on my tags and use them with my own spools. 

As i said in the first post, im willing to contribute to your research - got a flipper, chameleon tiny, chameleon ultra, px3 is on the way, also have 3 spools with tags - wich i can dump and share.... if neccessary -- but i guess not, as warbarz said "we have all the tags" (but dont want to share anything)  ;)

MatthiasWM, 2024-01-11:
I think the question is, if a published tag can be tracked back to the buyer, which is quite a privacy issue.

Sarah-1331, 2024-01-11:
I would doubt they record every tag 2per spool to the end user

rauschgiftengel, 2024-01-11:
So maybe we can make a list of the spools we have, and if some of us have
the same, we can decrypt the tags and compare the data. To see what
differences they have. I truly dont believe that they are traceable and
when there are unique numbers which allow, why should they? Have we broken
any law by decryping? Year... I'm my country it is.... But as i said, I
don't believe that. I'm happy to share my tags once my px3 arrives.

warbarz, 2024-01-11:
I have no idea where the tags I have, came from. Found them outside. Oh they were sold to a local business? Makes sense they would eventually discard old spools… after 200 or so of them… right?

warbarz, 2024-01-11:
So… having the tags physically, and dumping them are two different things… 

 

I have all of them, but havnt dumped them because there hasn’t been a good reason to invest the time. 

 

Those who have dumped the tags, have confirmed they are CRC locked, and we don’t have that encryption key yet to allow the ams to validate them.

 

So… correct me please, if I am wrong I want to know, but as it stands, though we can read and have deciphered the information in the tags, we have yet to reach the milestone of tag creation or manipulation, because, we cannot sign the tags crc after writing it. That encryption key is beyond cracking.

 

We are working on firmware and network at this point… I think tags are moot.

 

Sorry rauschgiftengel if that is unpleasant news… I hope to be wrong or a smarter person than I to come up with a solution. My proxmark is sitting idle and my storage room is filled with empty labeled spools with tags just waiting to be read so they can get reused/scrapped.

rauschgiftengel, 2024-01-11:
Okay, but wasn't mentioned in the discord that is possible to write the
complete dump to a compatible tag? Someone there wrote that it works and
the ams doesn't bother if there is the original and the cloned tag in it.

So in my understanding it is possible to dump and clone them, but it is not
possible to tamper with the data because of the locked checksum.

To not spam this issue, I would recommend to put this discussion back to
the discord :)

But Im really thankfull for your explanations and your invested time.

robertsamples, 2024-01-14:
Can you provide another discord link? I'd like to join but it is expired

Sarah-1331, 2024-01-14:
https://discord.gg/7bdDsUs3

mplacona, 2024-01-22:
> Okay, but wasn't mentioned in the discord that is possible to write the complete dump to a compatible tag? Someone there wrote that it works and the ams doesn't bother if there is the original and [...]

This is indeed true and I can confirm I've been able to get an existing tag information and write to my own tag and have the AMS recognise it.

@warbarz do you have dumps you can share since you seem to have them all? I'm particularly after White PLA to confirm my setup. 

Thanks

warbarz, 2024-01-22:
Yes-I can get that for you. I have yet to dump a single tag, I will need to familiarize myself with the process then will dump all tags for the group.Bambu white pla, no problem! Can someone assist me in discord with the process so I can get the right info produced? Then I will produce them all. I will post a request and look through the already posted process (seems last I looked to be a bit disjointed between two developers / scripts?) Ill figure it out.Note: I see some new spools on bambus site, and some old ones they have removed (sparkle pla for example). Last I ordered from them was october 2023. Since, I have been exclusively ordering sunlu, as they are bambus primary supplier (or so I am led to believe). I have all spools of all types and colors prior to oct '23.Sent from my Verizon, Samsung Galaxy smartphone

…

rauschgiftengel, 2024-01-23:
Good morning, what's your name in the discord? I can write an little guide
for you. An tag it to your name.

Wich hardware do you have or use for that? A pm3?

dhattey, 2024-01-23:
Please post another discord link. The one from 9 days ago has already expired.

Sarah-1331, 2024-01-23:
https://discord.gg/zJASwqup

Sarah-1331, 2024-01-30:
https://discord.gg/NzeZW2Fq

sanctas, 2024-02-06:
Can someone share a non-expiring link??

Sarah-1331, 2024-02-06:
Here never expire https://discord.gg/cfgpmhFn7S

lazerwolf22, 2024-03-29:
Discord link is invalid, please update. Thanks!