Bambu-Research-Group/RFID-Tag-Guide #19: Lookup table vs plain text

pciavald, 2024-05-05:
Hi,

I would like to give my opinion on the debate of using a lookup table or plain text as mentioned in the Open Spec document, also as to stir the debate so other people can give their opinions.

We have been active contributors of a few other related standards, including [Open Know How](https://standards.internetofproduction.org/pub/okh/release/1) which describes how to build something and [Open Know Where](https://standards.internetofproduction.org/pub/okw/release/2) which describes manufacturing capabilities.

When elaborating these standards, the decision was made to use wikipedia's article slugs to refer to existing tool types, manufacturing processes, materials and the such. That was convenient so that no specific actor has to maintain a database for others to access, which would increase centralization of the standard. The wikipedia slugs also have the advantage of being human readable, so if a slug is not present in a local database of slugs on a machine that is not connected to the internet, it will still be more readable than a sequential ID.

Materials in 3D printing have been known to have many names for similar stuff, as outlined in the Open Spec doc for PLA. However similar materials are often interchangeable in terms of printing process : PLA+, Tough PLA, PLA Silk will still be a PLA base offering different properties but will be printed with similar settings, being PLA base.

I propose the following:

The spec should include a plain text commercial name of the material identifying how the manufacturer sells it, and wikipedia slugs for the main component (>=51% of the composition) and for other components with optional percentages.

example:

```yml
name: Tough PLA-GF # required
type: polylactic_acid # optional wikipedia slug for PLA
components: fiber_glass:5,thermoplastic_polyurethane:3 # optional component that makes PLA tough with optional percentage in composition
```

Zsub, 2024-11-25:
I like the distinction between commercial name and actual material composition. Presumably it might be possible to define material families, that might serve all filament makers as well.

queengooborg, 2025-08-10:
Closing this issue as a decision had been reached to use plain text rather than lookup tables!

closed 2025-08-10 by queengooborg