Bambu-Research-Group/RFID-Tag-Guide #41: AMS bricks gen2 tags

Aptimex, 2024-10-09:
The Readme currently states that "Gen 2", aka CUID tags, "Work" for cloning. However, it seems the AMS (at least the AMS Lite on the latest firmware) tries to write invalid data to Block 0 to prevent the use of those tags, rendering them unusable in the process. Others have mentioned this as well https://github.com/Bambu-Research-Group/RFID-Tag-Guide/issues/1

**The Readme really needs to be updated with this caveat to prevent users from permanently bricking their Gen2 tags.**

capull0, 2024-10-09:
You can recover this tags with a PM3.
Or you pay much more per tag where you can change only one time the UID.
fun fact: the AMS will only brick RFID tags after reading the tag and recognize it as bambu spool tag :-)

After a while i came to the result, that cloning the RFID chips will help you with only some filaments, where the print profiles are near the same to Bambu filaments.
I found a reliable and similar way to tell the AMS which filament will be loaded.
https://github.com/capull0/BambuAMSCtrl
This is running on a raspberry pi with an external barcode reader and is working with any spools i have.
I can create my custom filament profile in Bambu Studio which will get a unique ID and with this tool you create a QR code just with this ID and the color.
Before you put a spool in the AMS, scan the QR code and and the tool will automatically select the correct filament to the empty AMS tray starting from left.
I'm working currently on a RFID variant, 2x PN532 connected to an ESP32

Aptimex, 2024-10-09:
Care to share the command to unbrick them? I was able to recover a blank/new one that I let the AMS Lite try to read (`hf mf wrbl --force --blk 0 -k FFFFFFFFFFFF -d 0102030404080400000000000000BEAF`), but the same command failed to recover one that I cloned before letting the AMS read it. Even after replacing the key value with the correct one from the clone. 

Yeah the one-time change tags should work fine, but the Readme REALLY needs to specify that. 

Neat project.

wzqvip, 2024-11-02:
Seems we should use FUID/UFUID, these tags only allow to change once or lock uid (forever).

thekakester, 2024-11-26:
I'm fine with updating the Readme.  It already said "gen 2" tags worked when I joined, and I didn't know what someone did to get them working.  I had the same thing where the gen2 tags would be "bricked", but I was able to unbrick them with some command that I'll try looking up.

thekakester, 2024-11-26:
Here's the guide I followed to unbrick: https://stackoverflow.com/a/65522498

wzqvip, 2024-11-27:
Is this machine specific? I get cuid work well on p1s. When I use this tag on x1c, it bricks.

I assume x1c and a1/a1mini had extra controllers(arm,esp32) to perform this task. P1 use the motion controller to operate the machine and the screen.

arnoliudaxia, 2024-11-27:
> Is this machine specific? I get cuid work well on p1s. When I use this tag on x1c, it bricks.
> 
> I assume x1c and a1/a1mini had extra controllers(arm,esp32) to perform this task. P1 use the motion controller to operate the machine and the screen.

I meet with exactly the same situation with you.

Aptimex, 2024-11-27:
> Here's the guide I followed to unbrick: https://stackoverflow.com/a/65522498

I was following that too, turns out I was just using the wrong key. Using the correct key dumped from the tag that was the source of the clone I was able to unbrick it. Thanks. 

Interesting that the cuid tags work on some machines and not others. I only have a A1with AMS Lite so I can't confirm behavior for any other setups.

MR-Ostrich13, 2024-11-27:
> You can recover this tags with a PM3.
> Or you pay much more per tag where you can change only one time the UID.
> fun fact: the AMS will only brick RFID tags after reading the tag and recognize it as [...]

Can you give me more information. It sound very interesting

MR-Ostrich13, 2025-01-02:
hello... now i am also an that point. my gen 2 chip was bricked by the AMS... all tehse commands down't work. even everone uses FFFFFFFFFFFF as Key...  but also the read Keys after and before cloning doesn'T work... has anyone an sollution? 

And additional to that is it possible to lock a gen2 afterwards. if not it dosn't make sense to go on with that

ESurge, 2025-01-18:
@MR-Ostrich13 

Bambu Lab wipes block 0 and adds a winky face ; ) to the first byte and thus "bricks" the tag. Cheeky.

To restore your Gen2 tag, do the following:
- View the dump of the original tag used for the clone, or rescan the original tag
- Use the 16 bytes (32 characters) of data on the Block 0 line and remove spaces for the write command
- Use Key B of Sector 0 for the write command (Last 6 bytes/12 characters of Block 3)

Example:
```
hf mf wrbl --blk 0 -k KeyB -d Block0_16Byte_Data --force -b
```

You may need to set some config parameters in PM3 in order to proceed with a broken BCC. I use the following:
```
hf 14a config --atqa force --bcc ignore --cl2 skip --rats skip
```
Afterwards, you should restore this config to default (or standard) which can be done with:
```
hf 14a config --std
```

Good luck.

lanrat, 2025-01-23:
I just had to reset one of my "bricked" gen2 cards. The steps listed in https://github.com/Bambu-Research-Group/RFID-Tag-Guide/issues/41#issuecomment-2599575688 worked for me, but I had to use KeyA instead of KeyB.

```
hf mf wrbl --blk 0 -k KeyA -d Block0_16Byte_Data --force -a
```

In case anyone else needs to do the same. The python code [snippit](https://github.com/Bambu-Research-Group/RFID-Tag-Guide?tab=readme-ov-file#key-derivation) that generates the keys using the KDF based on the card's UID can also be used to generate KeyA for each sector.  Alternatively you can determine the keys using the following proxmark3 script while running on the original tag that was cloned. It may take ~20min to run. 
```
script run fm11rf08s_recovery
```

In order to get retrieve `Block0_16Byte_Data` you can run the following on the original tag: 
```
hf mf rdbl --blk 0 -a -k KeyA
```

MR-Ostrich13, 2025-06-16:
@lanrat @exiom-xyz  thank you for that... i will try it. 

But is there any sollution to prevent the Brick or did someone found a cheap one time UID Gen 2 Chip???

exiom-xyz, 2025-06-16:
> [@lanrat] But is there any sollution to prevent the Brick or did someone found a cheap one time UID Gen 2 Chip???
The easiest solution I have found is FUID (Gen4 Write Once) cards, it is the most newbie friendly, both dumping and restoring tags are very straight forward.

[image]

Technically UFUID also works but requires extra steps to lock the card and I have bought ZUID which was sold as UFUID which will respond to Gen 1 (40) commands thus making them useless. Altho it is the same price for me, some seller sells these more expensively... however I don't really see the point for this specific application.

Aptimex, 2026-08-25:
Update: found [someone on Reddit](https://www.reddit.com/r/OpenBambu/comments/1vfkisc/bambu_rfid_writer_an_opensource_proxmark3_gui_for/) who bought a batch of CUID tags ([link](https://a.aliexpress.com/_Ez3pUXG)) that weren't being bricked by the AMS. I bought the same ones and confirmed that the AMS system doesn't brick them. (Disclaimer: there's no way to guarantee that the seller at that link will always ship the exact same tags, so YMMV)

Did some testing with my Proxmark and confirmed that these are safe to use because they actually enforce the ACLs on block 0, unlikely many other CUID tags. Bambu configures ACLs on their tags so that Keys A and B have read access to all blocks, while Key B also has write access to the ACL blocks. But many CUID tags ignore this for block 0 and allow Key A to write to it regardless of ACL bits. The ones linked above actually block that. So those tags ignore the write commands from the AMS and continue to function.

Here's a screenshot of the PM3 writing block 0 data to a vulnerable CUID tag (write succeeds), and then one of the tags linked above (write fails):
[image]

Both tags were first cloned using the same `hf mf restore --1k -u 1C1D5CB9` command, resulting in the same `878787` ACLs for all 16 sectors. The above write command was just attempting to re-write the same block 0 data that already existed there.

Gimno, 2026-08-25:
Hello,

What instructions did you follow?

dfrysinger, 2026-10-12:
Confirming @Aptimex's block-0 ACL finding on an **AMS 2 Pro** (P1S, AMS firmware 03.00.21.29), with results for 7 AliExpress listings.

**The test predicts AMS survival.** Clone a Bambu dump, then try to rewrite block 0 with its own bytes, using both keys:

```
hf mf wrbl --blk 0 --force -k <sector-0 Key A> -d <block 0>
hf mf wrbl --blk 0 -b --force -k <sector-0 Key B> -d <block 0>
```

`Write ( fail )` with both keys means the tag survives. `Write ( ok )` means the AMS will brick it. Writing back identical bytes is harmless either way. Every CUID tag below reports `Magic capabilities... Gen 2 / CUID` in `hf 14a info`, so only this behaviour test separates them.

A tag that refused the write showed up correctly in the AMS as ASA Black, and read back byte-identical to the library dump afterwards. On the AMS 2 Pro, block 0 was overwritten with the same fixed pattern every time: `99640020000000003E000000F3630208`, which has an invalid BCC.

Prices are what AliExpress US listed on 2026-10-12, with sale prices where a sale was on. They change often.

| AliExpress item | form | price (pack) | per tag | result |
|---|---|---|---|---|
| [1005009488943530](https://www.aliexpress.com/item/1005009488943530.html) | clear coin "CUID Gen2" (red coil; no adhesive, so it needs a holder or tape) | $20.06 / 50 | **$0.40** | ✅ survived the AMS 2 Pro |
| [1005006862472350](https://www.aliexpress.com/item/1005006862472350.html) | FUID keyfob (one-time UID; the chip is fragile, has no backing, and is glued into the fob, but can be removed with a heat gun and some effort) | $6.29 / 10 | $0.63 | ✅ survived the AMS 2 Pro |
| [1005008811881142](https://www.aliexpress.com/item/1005008811881142.html) | white coin sticker (the listing from #95 and the comment above) | $16.50 / 20 | $0.82 | ✅ survived the AMS 2 Pro |
| [1005010234960912](https://www.aliexpress.com/item/1005010234960912.html) | 5YOA wet-inlay sticker | $8.83 / 50 | $0.18 | ❌ bricked by the AMS 2 Pro |
| [1005010171868881](https://www.aliexpress.com/item/1005010171868881.html) | wet-inlay sticker | $2.42 / 10 | $0.24 | ❌ bricked by the AMS 2 Pro |
| [1005008229766738](https://www.aliexpress.com/item/1005008229766738.html) | sticker | $3.06 / 10 | $0.31 | ❌ bricked by the AMS 2 Pro |
| [1005008804142353](https://www.aliexpress.com/item/1005008804142353.html) | wet-CUID roll | $16.19 / 50 | $0.32 | ❌ bricked by the AMS 2 Pro |

Notes:
- The same store ("RFID Tag Card Store") sells two of the passing listings **and** one failing listing, so go by item number, not seller. Batches can still change, so run the test on every new batch before a tag goes near the AMS.
- The tags that refuse the write are still reusable. One went from PLA Matte Latte Brown to ASA Black (restore with the old keys, then the new dump) and still refused the block-0 write afterwards.
- On the passing tags, `hf 14a info` stopped reporting the Gen 2 / CUID magic once Bambu data was on them. I didn't check the failing ones after writing.
- Bricked tags can be recovered with ESurge's and lanrat's steps above (`hf 14a config --bcc ignore`, then write block 0 back with the sector-0 key).

Since this test reliably separates working from bricking Gen 2 tags, it might be worth adding to the README's "Gen 2 → Works" line.