Bambu-Research-Group/RFID-Tag-Guide #42: Filament Internal Code

spuder, 2024-10-23:
Dumping the mqtt messages I've identified an internal code that is used for filaments

[image]

WolfWithSword has documented most of these internal filament codes

https://github.com/WolfwithSword/ha-bambulab/blob/78cc6bda14ef0f82e74b59afc745cc5534c01a5e/custom_components/bambu_lab/pybambu/const.py#L40-L62

```
    "default": "Unknown",
    "GFU99": "Generic TPU",
    "GFS99": "Generic PVA",
    "GFL98": "Generic PLA-CF",
    "GFL99": "Generic PLA",
    "GFG99": "Generic PETG",
    "GFC99": "Generic PC",
    "GFN98": "Generic PA-CF",
    "GFN99": "Generic PA",
    "GFB98": "Generic ASA",
    "GFB99": "Generic ABS",
    "GFU01": "Bambu TPU 95A",
    "GFS00": "Bambu Support W",
    "GFS01": "Bambu Support G",
    "GFA01": "Bambu PLA Matte",
    "GFA00": "Bambu PLA Basic",
    "GFC00": "Bambu PC",
    "GFN03": "Bambu PA-CF",
    "GFB00": "Bambu ABS",
    "GFL01": "PolyTerra PLA",
    "GFL00": "PolyLite PLA"
}
```

CodeBradley, 2024-10-31:
You sir are a gentleman and a scholar. How are you dumping the mqtt messages like this btw? His script looked terminal based.

spuder, 2024-10-31:
To get the mqtt messages use mqttx

```
mqttx sub -t 'device/$SERIAL_NUMBER/report' -u -P $LAN_ACCESS_CODE --mqtt-version 3.1.1 -h $IP_ADDRESS -p 8883 -l mqtts --insecure
```

The mqtt api is documented here: 

https://github.com/Doridian/OpenBambuAPI/blob/main/mqtt.md

The 2 main endpoints are `/report` and `/request`

For example to change the filament color you would do a request with a json payload like this

```
{
    "print": {
        "sequence_id": "0",
        "command": "ams_filament_setting",
        "ams_id": 0,
        "tray_id": 0,
        "tray_info_idx": "GFG99",
        "tray_color": "F72323FF",
        "nozzle_temp_min": 220,
        "nozzle_temp_max": 270,
        "tray_type": "PETG",
        "setting_id": ""
    }
}
```

capull0, 2024-11-07:
you can query the Bambulab API and you will get a full filament list, also with your custom profiles.
each filament has a uniq ID, also the custom profiles.
You can have a look here, how you can read the filament list over the API: https://github.com/capull0/BambuAMSCtrl
I coded this flask application, which will make the Bambulab RFID tags obsolete.
Is working flawless since weeks.

johnhaxx, 2024-11-21:
> https://github.com/capull0/BambuAMSCtrl I coded this flask application, which will make the Bambulab RFID tags obsolete. Is working flawless since weeks.

Sounds amazing. I'll be following this pretty closely.

Elektroarzt, 2024-11-21:
> I coded this flask application, which will make the Bambulab RFID tags obsolete.

That sounds really cool! Is the following valid from your opinion to recognize custom filaments with the AMS?

Equipment
- Proxmark 3 easy RFID Reader
- compatible 13.56MHz RFID chip as described [in this issue](https://github.com/Bambu-Research-Group/RFID-Tag-Guide/issues/1#issuecomment-1701776114)
- Any computer running your flask app

Work Flow
1. Set new custom filament in Bambu Studio, generating a unique ID for it
2. Have RFID reader connected to computer running flask app
3. Flask app connects to Bambu Lab printer via API and MQTT
4. Scan RFID chip via flask app
5. App retrieves filament list with newly generated custom filament data
6. App connects RFID chip data with new custom filament ID and writes back to Bambu Lab printer
7. Stick RFID chip to custom filament spool
8. Printer will recognize the spool from now on with all data

Is that right or is something missing?
Thanks in advance for your help.

Elektroarzt, 2024-11-21:
This is absolutely amazing, you rock! 🤩
Have to try that asap. Will the RFID option utilize the onboard reader of the AMS to recognize inserted spools?

AndrewMohawk, 2024-12-07:
```
GFA00 - Bambu PLA Basic 
GFA01 - Bambu PLA Matte 
GFA02 - Bambu PLA Metal 
GFA03 - Bambu PLA Impact 
GFA05 - Bambu PLA Silk 
GFA07 - Bambu PLA Marble 
GFA08 - Bambu PLA Sparkle 
GFA09 - Bambu PLA Tough 
GFA11 - Bambu PLA Aero 
GFA12 - Bambu PLA Glow 
GFA13 - Bambu PLA Dynamic 
GFA15 - Bambu PLA Galaxy 
GFA50 - Bambu PLA-CF 
GFB00 - Bambu ABS 
GFB01 - Bambu ASA 
GFB02 - Bambu ASA-Aero 
GFB50 - Bambu ABS-GF 
GFB51 - Bambu ASA-CF 
GFB60 - PolyLite ABS 
GFB61 - PolyLite ASA 
GFB98 - Generic ASA 
GFB99 - Generic ABS 
GFC00 - Bambu PC 
GFC99 - Generic PC 
GFG00 - Bambu PETG Basic 
GFG01 - Bambu PETG Translucent 
GFG02 - Bambu PETG HF 
GFG50 - Bambu PETG-CF 
GFG60 - PolyLite PETG 
GFG96 - Generic PETG HF 
GFG97 - Generic PCTG 
GFG98 - Generic PETG-CF 
GFG99 - Generic PETG 
GFL00 - PolyLite PLA 
GFL01 - PolyTerra PLA 
GFL03 - eSUN PLA+ 
GFL04 - Overture PLA 
GFL05 - Overture Matte PLA 
GFL06 - Fiberon PETG-ESD 
GFL50 - Fiberon PA6-CF 
GFL51 - Fiberon PA6-GF 
GFL52 - Fiberon PA12-CF 
GFL53 - Fiberon PA612-CF 
GFL54 - Fiberon PET-CF 
GFL55 - Fiberon PETG-rCF 
GFL95 - Generic PLA High Speed 
GFL96 - Generic PLA Silk 
GFL98 - Generic PLA-CF 
GFL99 - Generic PLA 
GFN03 - Bambu PA-CF 
GFN04 - Bambu PAHT-CF 
GFN05 - Bambu PA6-CF 
GFN06 - Bambu PPA-CF 
GFN08 - Bambu PA6-GF 
GFN96 - Generic PPA-GF 
GFN97 - Generic PPA-CF 
GFN98 - Generic PA-CF
GFN98 - Generic PA-CF  P1P
GFN99 - Generic PA
GFN99 - Generic PA  P1P
GFP95 - Generic PP-GF 
GFP96 - Generic PP-CF 
GFP97 - Generic PP 
GFP98 - Generic PE-CF 
GFP99 - Generic PE 
GFR98 - Generic PHA 
GFR99 - Generic EVA 
GFS00 - Bambu Support W 
GFS01 - Bambu Support G 
GFS02 - Bambu Support For PLA 
GFS03 - Bambu Support For PA/PET 
GFS04 - Bambu PVA 
GFS05 - Bambu Support For PLA/PETG 
GFS06 - Bambu Support for ABS 
GFS97 - Generic BVOH 
GFS98 - Generic HIPS 
GFS99 - Generic PVA 
GFT01 - Bambu PET-CF 
GFT02 - Bambu PPS-CF 
GFT97 - Generic PPS 
GFT98 - Generic PPS-CF 
GFU00 - Bambu TPU 95A HF 
GFU01 - Bambu TPU 95A 
GFU02 - Bambu TPU for AMS 
GFU98 - Generic TPU for AMS 
GFU99 - Generic TPU
GFU99 - Generic TPU  P1P
```

Do these match your already known IDs?

pelrun, 2024-12-07:
FYI BambuStudio has all the filament data in .json format including the filament_ids, so you don't have to jump through hoops to get the latest list (note, only the *base.json files have the id): https://github.com/bambulab/BambuStudio/tree/master/resources/profiles/BBL/filament

CodeBradley, 2025-01-09:
> > > I coded this flask application, which will make the Bambulab RFID tags obsolete.
> > 
> > 
> > That sounds really cool! Is the following valid from your opinion to recognize custom filaments with the AMS?
> > Equipment
> > 
> > * Proxmark 3 easy RFID Reader
> > * compatible 13.56MHz RFID chip as described [in this issue](https://github.com/Bambu-Research-Group/RFID-Tag-Guide/issues/1#issuecomment-1701776114)
> > * Any computer running your flask app
> > 
> > Work Flow
> > 
> > 1. Set new custom filament in Bambu Studio, generating a unique ID for it
> > 2. Have RFID reader connected to computer running flask app
> > 3. Flask app connects to Bambu Lab printer via API and MQTT
> > 4. Scan RFID chip via flask app
> > 5. App retrieves filament list with newly generated custom filament data
> > 6. App connects RFID chip data with new custom filament ID and writes back to Bambu Lab printer
> > 7. Stick RFID chip to custom filament spool
> > 8. Printer will recognize the spool from now on with all data
> > 
> > Is that right or is something missing? Thanks in advance for your help.
> 
> No, it's much easier. Right now i'm using a barcode reader, the module for reading RFID chips is not yet implemented. Since only the unique Filament ID and color is read from the QR-code or from the RFID tag (when implemented), you don't need a Proxmark3 and you can use any RFID tag. At the QR-code or the RFID tag (when implemented) is only the UUID and the color-code stored, everything else is configured in your Bambu Studio filament profiles.
> 
> Work Flow: prepare a filament spool:
> 
> 1. Open the Website, the Flask App offers and login to your Bambulab Cloud Account (LAN only not yet tested, but should also work). The generated token is stored and used for all requests.
> 2. Open the Filament page, the app is fetching all filaments and sort all builtin and custom filament profiles.
> 3. select filament and the color. Either apply this settings to your printer directly or print the generated QR-code and place it on the filament spool (later there will be a button to store the information on a RFID tag)
> 
> print with this filament:
> 
> 1. The app is reading over MQTT the AMS and is getting all empty AMS slots.
> 2. If you scan a QR code (or later the RFID tag) the App will set this filament to the most left empty slot.
> 3. Load the filament spool to the most left empty slot and you will see the selected filament in the device tab of Bambu Studio.
> 4. Press the sync button in your prepare tab.
> 5. Print

I'm working on a project that uses a color sensing module to scan in all of my filaments inside of a drawer system I setup, categorized by material, everytime I   tap a button. An ESP32 will send the colors of each spool to an RFID chip that will program a blank RFID sticker, the type Bambu uses. 

It sounds like I could use this to read those colors even though it's not a Bambu sticker. I'm guessing this just intercepts the mqtt requests that the AMS sends on read and sets it via an API? I'm trying to envision how a QR code on the filament would be read and how it would write to RFID in your setup, I'm guessing that would need to use custom hardware? 

I'll probably edit this once I can skim the code.

Elektroarzt, 2025-01-09:
Will you make this project accessible on Git? Sounds interesting

commit c8042761 (2024-12-06): Add known material IDs

Fixes #42.
closed 2025-06-14 by thekakester