Bambu-Research-Group/RFID-Tag-Guide #43: OpenSourceRFID support NDEF tags

spuder, 2024-11-17:
The current [RFID tags](https://github.com/Bambu-Research-Group/RFID-Tag-Guide/blob/main/OpenSourceRfid.md#rfid-hardware) for OpenSourceRFID call for MiFare Classic 1K tags

While these tags are cheap and ubiquitous; they do not support NFC Forum Type 2 NDEF records

## Proposal for change

OpenSourceRFID should transition from `MiFare Classic 1K` to either
- `MiFare ultralight`

or 

- `NTAG215/216`

## Reasoning

There are 3 stakeholders in the OpenSourceRFID protocol
1. Filament Manufactures
2. 3D Printer Manufactures & Microcontroller based RFID readers
3. End users

Stakeholders 1 & 2 are well served with MiFare Classic 1K. 
Stakeholder 3 would be better served with an interface that allows reading RFID tags with a cellphone. 

### Example Use Case

End user picks up filament spool, touches their phone to the tag, and is able to view metadata about the spool (hex color, temperature ranges, manufacture date, ect...). 
The user benefits by being able to view information without requiring access to specialized RFID reading hardware (RFID Reader, 3d printer ect...)

### Demo

[NFC Phone Example - Youtube](https://youtube.com/shorts/pWnvJc-8KLI?si=9-C1rhqrvL1i5CpR)

[image]

## Technical information

In the above proof of concept, a single `http://` ndef record is written to the NFC tag. (For IOS users it's essential that the NDEF record be record 0). The ndef record provides query string parameters to a website. The NFC tag contains as little information as possible and defers the presentation/business logic to javascript on the website. 

```
https://openspool.io/tag_info?color_hex=FFAABB&brand=Generic&min_temp=220&max_temp=240&protocol=openspool&version=1.0
```
Here the only required information is

- https://openspool.io/tag_info
- color_hex
- brand
- min_temp
- max_temp
- protocol
- version

[Example javascript](https://github.com/spuder/OpenSpool/blob/16737e1e738be204b0075b86e6249ae03ab52db6/docs/tag.md#L259-L291)

### Considerations

**Cost differences**:
A quick search of aliexpress/amazon found Ultralight and NTAG 215 for _less_ money than Classics (in small batches). Production costs would likely be very different. 

| | | 
| --- | --- | 
| Mifare ultralight  | $0.10  | 
| MiFare Classic |  $0.25 | 
| NTag 215 |  $0.11 | 

**Memory Size**
NTag 216 has a size of 888bytes. While smaller than the Classic, 888 bytes should be sufficient. 

By comparison, the OpenSpool protocol uses NTAG 215 with 500 bytes tags  and has yet to exceed 300 bytes in any real world situation.

### Call to action

Criticisms, observations, rebuttals, opinions welcome. I invite the community to weigh on on this proposal.

BoKKeR, 2024-11-26:
Shouldnt this be up to the end user what kind of NFC tag/reader they end up using?

queengooborg, 2025-08-10:
Closing this issue since a decision was reached to use NTAG tags!