Bambu-Research-Group/RFID-Tag-Guide #44: Use tagged code not memory map (just a suggestion)

DanBGold, 2024-11-21:
This is just my opinion and it can be ignored but if you want to make the code future proof it should be a readable tagged code not memory mapped. To make it as small as possible it should just be key value pairs. Such as
VER:A1;MAN:eSun;BRD:eSun;TYP:PLA;STP:Plus;COL:Fire Engine Red;DIA:1.750;WGT:1000;PTP:220;BTP:60;DST:1.23;
: colon and ; semicolon are reserved characters
All white space (space, tab) is converted to 1 space.
VER version
MAN manufacturer (eSun, Creality, etc)
BRD brand (for manufacturers with multiple brands such as Creality has Ender, Creality, CR and HP)
TYP type (PLA, PETG, ABS, ASA, TPU)
STP sub type (Plus, Hyper, Wood, Carbon, Silk, 95A, 85A)
COL color (Red, White, Blue, green) must have at least one COL, RGB or CMY tag
RGB RGB color
CMY cmyk color
DIA diameter in mm (include decimal)
WGT weight, not sure if this should be mandatory, spool weight probably should be as the tag is on the spool and having a scale on the printer would give exact weight if the spool is subtracted.
PTP suggested print temperature
BTP suggested bed temperature
DST density
Shouldn’t have data that is dependent on printer or environment. The goal would be a printer firmware should be able to have multiple tables of slicing parameters for a filament on a  machine in an environment and convert the gcode to another filament with other parameters.

thekakester, 2024-11-26:
We talked about this previously, and would likely run out of memory very quickly.  NFC tags only have about 800 bytes of memory to work with, and 256bytes will likely be taken up by a signature (which is something filament companies seem to be rather insistent on)

For example, storing "WGT:1000" uses up 9 bytes where it would otherwise use 2 bytes if memory mapped.
This should ideally make parsing the tag easier for microcontrollers too because doing string parsing uses more memory + computational power.

DanBGold, 2024-12-21:
The tags would need to be limited in size maybe only 4 characters plus a termination character (tab 0x09) or set a specific number of character's so no end is needed. The data could then be ended by a single character (\n Ox0A).  It could even be made smaller by defining the data by the tag and then binary values can be used instead of characters, this would limit the human readable part. Having the tag instead of defining every bit of data keeps it expandable. 

Also did some more research on how the  Creality CFS is able to store filament left as it only has access to one RFID so it can't modify both. As far as I can tell it does not store the amount of filament left. It calculates it by reading the tag feeding filament till it seas the tag again and determines how much filament is left by the amount fed as this would be the circumference of the outer layer. So the type of spool for the dimensions or the dimensions of the core and internal width would be valuable. As the Bamboo has the same limitations I would think they do the same.

queengooborg, 2025-08-10:
I'm going to go ahead and close this issue as the final decision was to use memory mapping, and in doing so, the core data was able to easily fit on a 144-byte NTAG213.

closed 2025-08-10 by queengooborg