Bambu-Research-Group/RFID-Tag-Guide #45: Repurposing BL Tags

bryan-lott, 2024-11-23:
Apologies for any dumb questions or misunderstandings, I'm just now getting into how NFC tags work and I haven't been able to find information on repurposing the tags for things other than new spools. I.e., if I want to take one of the tags and use it as the insert in an NFC "Business Card" or as a "tap to get the wifi password" thing.

Obviously, I can just go buy NFC tags, but it'd be great to be able to reuse the ones from the filament rolls.

Is there any way to force a "format" of the tag and/or overwrite it with different information for a different purpose?

Thanks!

dequally, 2024-11-24:
You probably cannot write over the tags (I have not tried), since the manufacturer can make parts or the whole RFID tag read only.

For more information, I found this 
https://stackoverflow.com/questions/501302/can-i-write-or-modify-data-on-an-rfid-tag

PS. I would consider this off topic, as it is not specifically related to this repository or project. Consider closing the issue

thekakester, 2024-11-26:
Yes, bambu tags are writable.  These specific tags are MiFare, which are not the same as NFC (your phone won't be able to read mifare).  There are two sets of keys for bambu tags.  A, and B.  If you know the A keys, it allows you to READ from the tags.  If you know the B keys, it allows you to WRITE to them.  Both have been discovered, but B keys have been intentionally not documented because writing over bambu tags seems like it could cause more harm than help.  For example, someone with a flipper could go around "erasing" the tags.  The details on the A keys which allow tags to be read have been documented.

bryan-lott, 2024-11-26:
That's great context and info, really appreciate you taking the time!

closed 2024-11-26 by bryan-lott