Bambu-Research-Group/RFID-Tag-Guide #48: Extended Color Hex Format

wherron01, 2024-11-24:
This is a simple proposal to extend the color hex field in the open source RFID standard. I have four proposals for how this could be done, in increasing order of complexity. 

A basic implementation would keep a single color, but use 4 bytes in order to include an alpha channel. As both translucent filaments and RGBA values are fairly common, this would be a simple change that could support more filaments while adding only a single additional byte to the format. 

Alpha could also be used in non-standard ways to represent things like twinkling filaments, by dividing its byte in half or more, or using a lookup table (though this was to be avoided generally, it does provide unique benefits). Filament translucency should never be more granular than a single hexadecimal digit, so using 4 bits to represent transparency and another four to represent other properties would be a simple solution that reduces size while maintaining the ability to display things like sparkle. To avoid a lookup table, this could also represent different boolean properties with each of the bits.  For example, four bits could be allocated to alpha, while the other four represent matte vs gloss, twinkle vs standard, metallic, and color shift vs monochrome. 

Color shift brings me to the next proposal. Many filaments contain more than one color, and a single hex code is unable to represent their color adequately. To accommodate this, the bits initially allocated to alpha or properties could be used to store a color quantity. For example, in what was initially proposed to be the “alpha” channel, the first four bits could be allocated to an index and the other four for a quantity. If that number was three, eg the color hex code was XX XX XX 02, this would be followed by two more color hex codes. For example, a red-green-blue color shift filament would be represented by FF00000200FF00120000FF22, or (FF 00 00 02) (00 FF 00 12) (00 00 FF 22).  This provides basic support for color shift filament, but sacrifices support for visual properties. 

The final proposal, though the most expensive, sacrifices nothing. This would be to use the first byte to encode quantity, with the first half of the byte encoding channels per color and the second half encoding colors. for example, 20 XX XX XX would be equivalent to the current situation, with three bytes per color and a single color, while only taking up one extra byte, equivalent to the alpha channel proposal. However, this is extensible up to FF…, which would permit 16 colors with 16 bytes each, for a huge number of visual properties per color and a huge range of color shift per filament. A 50% alpha version of the RGB color shift filament described above, for example, could be represented by 32FF00007000FF00700000FF70, or 32 (FF 00 00 70) (00 FF 00 70) (00 00 FF 70). 

The dynamic size of the final two proposals could present problems, but given the size is presented in a fixed location, it would still be heuristically parsable. It could also be the final parameter (given it’s already the second-to-last), which would simplify things.

thekakester, 2024-11-24:
This is a good point.  Representing rainbow, dial-color, and tri-color
isn’t something this does yet.  We’ll also probably want to plan ahead for
something that hasn’t been invented yet.  I’ll give this some thought.  If
you’re on the discord server, this might be a good thing to bring up to get
opinions from everyone else.

wherron01, 2024-11-24:
I am not on the discord.  I see that it's only been referenced in other issues, and the invite in another issue is broken.  Perhaps a perpetual invite in the readme would be good?  Either way, an invite to me would be appreciated

thekakester, 2024-11-26:
An invite has been added to the readme.  I wasn't sure if that was something that the original discord owner was ok with, so I had to check first.

queengooborg, 2024-12-05:
The link in the read me actually got removed on accident, here's the link in the meantime: https://discord.gg/zVfCVubwr7

mytechguyri, 2025-01-30:
I'd recommend sticking to just adding alpha channel for its intended purpose, i.e. translucent filaments.. otherwise, you're deviating too much from the existing RRGGBBAA hex color standard

BenBE, 2025-01-30:
Recently I stumbled upon [this site](https://spectraldb.com/), which collects color measurements for various materials and surfaces. Depending on the filament type this is severe overkill, but things like viewing angle-variant filaments and color-changing could be expressed as a multi-parameter function returning a 4+ dimensional value:

Dimension 1-4*: RGBU color as normal
Dimension 0: Transparency
Dimension -1: Surface Sparkling
Dimension -2: Surface Specularity
Dimension -3: Surface Roughness
…

The input parameters for this function would then be:
Parameter 1: Offset on spool
Parameter 2 and 3: Viewing Angle
Parameter 4: Temperature
Parameter 5: Humidity
…

The color definition would then need to define the output for this function with various inputs, where intermediate values are interpolated between nearest hyperplane neightbours.

A filament that starts red and goes to green to the end of the spool and emits green light under UV could then be something like:

```
offset=0 : transparency=0.02 sparkling=0 specularity=0.05 roughness=0.8 @780=(t=0.8, r=0.95, e=0.0) @520=(t=0.25, r=0.2, e=0.4) @400=(t=0.05, r=0.05, e=0.0) @320=(t=0.2, r=0.1, e=0.0)
offset=1 : transparency=0.02 sparkling=0 specularity=0.05 roughness=0.8 @780=(t=0.2, r=0.35, e=0.0) @520=(t=0.85, r=0.8, e=0.4) @400=(t=0.05, r=0.05, e=0.0) @320=(t=0.2, r=0.1, e=0.0)
```

A rainbow filament would just use more offsets for intermediate colors. For color-changing filaments based on temperature you'd define the colors based on temperature, etc. Yes, that's kinda verbose, but allows for the most flexibility when new parameters are added.

One aspect that would need to be discussed with this format is frequency-dependent emission spectra, to capture how the colors would look in different lighting conditions, but this could be modeled simply by specifying the function with the wavelength as an input parameter too (otherwise "white" light is assumed).

*Actually some wave lengths, to allow for different levels of details with the reflectivity/emission. Color dimensions would need to come in tuples for transmission, reflection and emission.