Bambu-Research-Group/RFID-Tag-Guide #54: How to clone existing tags? Compiling information.

token47, 2024-11-28:
This project is awesome and I think it already got very far being able to read tags contents. But because of the signature, it seems like it is still a bit far away the possibility of actually creating your own tags from scratch.

But something that seems to already be possible is to CLONE existing tags, including their signature. I have seen bits of information about this scattered all over the place, but nowhere I can find a very direct and to the point tutorial of cloning existing devices.

What should be compiled:

- a directory/repository of all existing tags on the market today (all colors, all filament types). There's probably no need for repeated ones where only the UUID and signature changes, but having a directory where you can choose one that matches your 3rd party filament and clone it to a new tag would be very convenient (there are samples and examples floating around but only a few, I have not seen a proper comprehensive directory). This could be hosted in a website somewhere or added to a directory on the project.
- a list of compatible (and tested) tags and where they can be purchased to make everyone's life easier.
- a tutorial explaining what is the equipment needed and how to download one of the tag dumps from the directory and how to write them to a new (blank) tag.

This is something totally possible today and would solve the biggest need from users which is making 3rd party filament spools as convenient as the originals. Of course there are other needs but this would already go a long way.

I hope that this issue will help to organize all the info needed, focusing exclusively on cloning.

thekakester, 2024-11-28:
I did this a while ago and made a video showing it working (https://www.tiktok.com/@polarfilament/video/7375211374751780138?lang=en), but I haven't had a chance to make a TUTORIAL video on it yet.

thekakester, 2024-11-28:
Just to document it here, the details are in the tag choice, and learning how to debug via the tracelog captured using a proxmark3 between the tag and the AMS.

The first thing that happens is the AMS reads the entire tag (including signature) and verifies that it's correct.  If it IS correct, they try to brick the tag by writing to Block 0.  Block 0 is normally read-only, so this command does nothing to standard RFID tags (eg genuine bambu tags).

For magic tags (such as magic gen2), Block 0 is writable, therefore the command succeeds and writes garbage data to the tag, therefore "bricking" it.  This is their defense against people using Magic tags.

However, there's such things as FUID (fused UID tags) which allow block 0 to be written to ONCE, but never again.  This lets you clone a bambu tag, and then when the printer tries to overwrite block 0, it fails.  This convinces the printer that the tag is genuine, and it reads it correctly.

Other than that, it's just a standard ProxMark3 write instruction to write to your FUID tag.  The complexity is just debugging in the event that it doesn't work.

token47, 2024-11-28:
This other issue is mapping compatible tags: https://github.com/Bambu-Research-Group/RFID-Tag-Guide/issues/1

MR-Ostrich13, 2024-11-29:
Hey this is exactly the thing what I am doing in the moment. 
So I will support with red tags and would extend the list with trying out some filaments with these settings. Which work good and which not. 

So where do we place this collection?

token47, 2024-12-13:
Ok, here are a few questions:

- when I'm using a magic card that the printer "bricks", does it get unusable or it just gets garbage and I can program it again?
- considering I have a dump file (let's say the one in the example dir) what is the exact command to write it to a card?
- is the command different if the card is a FUID or not? According to some reading it seems like magic gen1 uses cload (using backdoor commands) while the fuid should use wrbl (treated like gen2, writing directly to block 0).

Stayready83, 2024-12-31:
Annnnnnnnddddd no one replied …….. frustrating. You were almost there. I tried to clone some Creality tags using an rfid reader writer to no avail. I think us cloning these tags is a big deal because if we can easily do it and get that info out to the public so everyone is doing it. It renders their little locked system useless possibly more encouragement for them to join the open source project I would think. Why wouldn’t they at this point….. once they realize their tags are readable writable basically they no longer control where you but your filament. That’s the biggest deal. They want you to buy their filament.

token47, 2024-12-31:
Yeah, I got the equipment and some tags, including normal and FUID ones, I'm just not sure how to proceed.

raihei, 2025-01-03:
> Yeah, I got the equipment and some tags, including normal (magic) and FUID ones, I'm just not sure how to proceed.

If you have some genuine Bambu tags/spools, I would start with reading the data off of these. Then you have some dumps you can try to restore to the gen2 tags. How to read and restore data from/to Mifare Classic tags should be an easy find via Google.

I think, that the bricked tags are not recoverable. I only have a PN532 so I don't know if a Proxmark is mayble able to recover them, but I think not. I read somewhere, that when there is invalid data in a sector the sector is "dead" forever.

MR-Ostrich13, 2025-01-08:
ok i also wanne go on with that and still searching gen3 or FUID (Gen2 OTP) tags. @token47 you've said you have some. could you write me which ones and where to get thes. (As Stickers not as card)

wesbos, 2025-03-25:
Has anyone shared their tags data anywhere? I want to get into this but would need a resource for cloning tags that I don't have

bbbenji, 2025-03-25:
@wesbos hop into the Discord, you can find the link in the [readme](https://github.com/Bambu-Research-Group/RFID-Tag-Guide/tree/readme) branch. There's a bin-dump channel there.