Bambu-Research-Group/RFID-Tag-Guide #60: Programming an FUID  Tags-> the working procedure

MR-Ostrich13, 2024-12-24:
Hello
i read some diffrent ways to copy / clone a chip.... so i wanne make an overview of working procedures. could you help me. 
thinking of that

1. Tool: (Android App, Proxmark ...)
2. commands: (restore .....)
3. which tag: 
4. result: (Works just one time, works every time, again and again...)

would be grat to have an overview of working steps and results

pruwait, 2025-06-07:
Read bambu tag just few seconds with Proxmark3.

```
hf mf bambukeys -r -d
```
```
hf mf dump
```

pruwait, 2025-06-07:
Find bambu tag in closed spool box with proxmark3.
`hf tune `
it must be less 5700 for good reading.

exiom-xyz, 2025-06-23:
> Hey Reading and dumping still works. But the AMS destroy still the chip. How could I prevent that ?

Per my testing, I only have the H2D, AMS 2 Pro and AMS HT, but I assume with firmware updates, it would be the same for all printers if not already then eventually... it would seem Gen 2 tags (CUID) no longer works and you will have to use FUID or UFUID tags