Bambu-Research-Group/RFID-Tag-Guide #62: Idea of an Android app to display spool info

thomasfla, 2025-01-07:
Hi, thanks for the nice reverse engineering, 

We have a lot of bamboo spool and refill them ourselves. It happened that we forget to glue the new label.

I did a simple test with my andoid phone, reading a UID with MIFARE Classic Tool app, then running the key generator python code provided here on my PC, writing the keys to a key file back on my phone, and now I can read and decode a tag with it:
[image]

It would be cool to create an android app to automate this and parse the spool ! WDYT? 
I might give it a try if it makes sense..

thomasfla, 2025-01-08:
I did a POC : 
https://github.com/thomasfla/3DSpoolScanner
[image]

Can try it here:
https://github.com/thomasfla/3DSpoolScanner/releases/download/untagged-6c3b72bc41ad535b3964/3dSpoolScanner.apk

mrambossek, 2025-01-08:
this looks really cool. i have zero experience with android development - is it feasable to run the key generation code right inside the app?

thomasfla, 2025-01-08:
Hi, yes, this is what I did. 
I have also zero experience with android, but I've spent an afternoon with ChatGPT...
The app scan the UID, derives the key and read block 4 and 5 right now.

Amerzel, 2025-01-14:
> I did a POC : https://github.com/thomasfla/3DSpoolScanner ![1736348969028](https://private-user-images.githubusercontent.com/11156435/401202184-df7abeb9-fe4a-4cc9-9a98-314080f3e7e5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4ODA0ODMsIm5iZiI6MTczNjg4MDE4MywicGF0aCI6Ii8xMTE1NjQzNS80MDEyMDIxODQtZGY3YWJlYjktZmU0YS00Y2M5LTlhOTgtMzE0MDgwZjNlN2U1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTE0VDE4NDMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjYTA3ZGMxODI0Y2Q5M2QxZDZhM2E3NDIzNDM5ZjUwNmRlNzViYjg1MGJlY2RiOTEzMGZkZTRlMzcyYjZiNWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Y_egbMDhtkSaOrKJ91yvTDj2Q-TOLzHoIC8ZdWxAXao)
> 
> Can try it here: https://github.com/thomasfla/3DSpoolScanner/releases/download/v0.1/3dSpoolScanner.apk

This is awesome, worked great. Nice first steps!

cherryduck, 2025-01-18:
Bit of duplication, I've also created an android application which does the same but also dumps the whole tag, and exports the dump and the key files. Feel free to try it out!

https://github.com/cherryduck/BambuTagScanner

MrB-PrintingFun, 2025-03-09:
Hello, based on cheryduck's excellent work, I've made a proxy in python to fill SpoolMan database
RFID scan through phone -> API call to my proxy -> complete decoding -> colors and manufacturer infos from spoolman DB -> spoolman calls to add spool

If interested in, I'll publish it soon

MrBambuSpoolPal, 2025-04-08:
Hello, here is a release candidate: 
https://github.com/MrBambuSpoolPal/MrBambuSpoolPal-BambuSpoolPal_AndroidApp/tree/bst_v1_0/releases

Documentation is on the way:
- spoolman withing docker with HTTPS: https://github.com/MrBambuSpoolPal/MrBambuSpoolPal-BambuSpoolPal_AndroidApp/tree/main/docker

[image]