Bambu-Research-Group/RFID-Tag-Guide #72: How to write.

ibrahimxq, 2025-04-09:
I want to use the RFID tag to make another thing because I have too many, but I can't understand how to write on them.

niyawe, 2025-04-19:
Step One:
Replace `RFID-A\0` with `RFID-B\0` in HKDF to generate B-Keys

Step Two:
Replace `878787` with `FF0780` in the last Block of every Sector (resets permissions)
Or: Write `000000000000FF078069000000000000` to the last Block of every Sector (resets permissions and sets A- and B-Keys to `000000000000`)

Step Three:
Use the A- and B-Keys or the newly set Keys to write to the other Blocks

niyawe, 2025-04-19:
I'm refering to the Key Derivation Script https://github.com/Bambu-Research-Group/RFID-Tag-Guide#key-derivation

ibrahimxq, 2025-04-19:
ok so the readme file has all the things to write on it? sorry because I need to learn rfid and nfc.

ibrahimxq, 2025-04-24:
hello i am using the key derivation code and have been able to generate keys, now how can I save them to a bin file because I don't know

closed 2025-05-07 by ibrahimxq