Bambu-Research-Group/RFID-Tag-Guide #73: Typo in the 'RFID-Tag-Guide'

sepp89117, 2025-05-13:
The 'RFID-Tag-Guide' specifies the length of 'Tray Info Index - Unique Material Identifier' (Block 1) as 16 bytes. According to example data, however, the length is 8 bytes.

sepp89117, 2025-05-17:
By the way, I find the term "Unique Material Identifier" misleading. It's simply the ID of the specific material. So, every PLA Basic has "GFA00," every PLA Matte has "GFA01," and so on. See the 'filament 0.x.json' files, where it's called "filament_id." Unique, to me, is a one-time random identifier.

In the Filament JSON files, you'll also find "setting_id," which is very similar to block 1, index 0, which you call "Material Variant Identifier." If I had more tag dumps, I could confirm that "Material Variant Identifier" is indeed "Setting ID." Can you verify this?

Filament JSON files from the offline ota firmware file for the X1:
[filament-0.2.json](https://github.com/user-attachments/files/20261288/filament-0.2.json)
[filament-0.4.json](https://github.com/user-attachments/files/20261291/filament-0.4.json)
[filament-0.6.json](https://github.com/user-attachments/files/20261290/filament-0.6.json)
[filament-0.8.json](https://github.com/user-attachments/files/20261289/filament-0.8.json)

queengooborg, 2025-06-24:
This had been fixed in #55!

closed 2025-06-24 by queengooborg